gatechipdl/BirdAttractor #1: design basic form for board components and case

strontium5, 2016-09-13:
arduino shield form
pin locations
power plug-ins
battery changing (every 8 years)

racheleleroy, 2016-10-31:
this isn't for the the 3D form, but for the UI. Couldn't figure out how to upload new file, kept saying I didn't have that authorization. 

Very basic UI, still working on learning html, css, js but wanted to get some basic skeleton for mvp after doing the tutorials to exercise the new things in my head.
[bird attractor device rach's comp.zip](https://github.com/gatechipdl/BirdAttractor/files/561059/bird.attractor.device.rach.s.comp.zip)

strontium5, 2016-10-31:
this is for the 3D form

strontium5, 2016-10-31:
You should be able to commit now.
Also, I added issues related to the UI. You can change them as you see fit.

racheleleroy, 2016-11-12:
3D files for the form to hold the board and components. Built in fusion360. The two stl files are each half of the box (top and bottom) separate so that they can easily be processed in the design shop. The other files are a variety of 3d files for good measure. 

[3D form v1.zip](https://github.com/gatechipdl/BirdAttractor/files/587514/3D.form.v1.zip)

strontium5, 2016-11-30:
should be moved into the github code section under models ( I will create and do that now)

strontium5, 2016-11-30:
uploaded.
you can use the RevC pcb layout pdf file to get mounting hole and component positions for the next version of the case:
https://github.com/gatechipdl/BirdAttractor/blob/master/Hardware/BirdAttractor_RevC/BirdAttractor_RevC-Layout.pdf

strontium5, 2016-12-05:
@racheleleroy I added a photo of the assembled Rev C board. There are a few issues, but it should work. I haven't started on the basic firmware yet, but I did add the issues to the github issues list. I'll leave this board in our project box in the IPDL.

racheleleroy, 2016-12-06:
@strontium thanks for moving it to models and for the photo. Ill stop by and take a look at the board. I can have v2 by end of weekend. I'll slip it onto the design shop machines while I work this weekend