gatechipdl/BirdAttractor #5: design overall UI/UX for all pages hosted on device

strontium5, 2016-10-31:


racheleleroy, 2016-11-28:
[bird.att.take.two.zip](https://github.com/gatechipdl/BirdAttractor/files/615287/bird.att.take.two.zip)
Work on the bootstrap UI. No specific UI for picking date and time yet. There are some plugins we can use - going by the app lab tomorrow (monday) to get help understanding how to set them up across the html, css, and js

strontium5, 2016-11-30:
need to remove/replace references to external sites, such as for javascript libraries. This will not be connected to the internet. 
This line is a problem:
<script src="//ajax.googleapis.com/ajax/libs/jquery/1.11.2/jquery.min.js"></script>
There is a local version of jquery in the js folder, which should be referenced instead.

It is looking alright so far. May need to distinguish between first time users, and those who just want to go and set the time.

While this device is being developed for the bird attractor, it is really much more generic than that. It is a device for setting 5v relay alarms over wifi. Should we make a distinction in the text (ie scientist vs user)?

strontium5, 2016-11-30:
Also, you can just put your code into the github. You won't need to create different version names for this code either, as github will maintain every version. I can put it in if you want. But eventually you should learn how to use git.